pasqal-io/Pulser #66: Integrating ReviewNB

HGSilveri, 2021-01-15:
Now that we are open-source, we can integrate [ReviewNB](https://www.reviewnb.com/) for free to facilitate the review of Jupyter notebooks. What do you think, @lhenriet  @sebgrijalva ? Should I set this up?

sebgrijalva, 2021-01-15:
I think it's a good idea, we might as well get used to it now, as the number of notebooks will keep growing.

HGSilveri, 2021-01-18:
I know I said I'd do this, but actually it has to be done with the account that owns the repository, so I think it will be up to you, @lhenriet .

lhenriet, 2021-01-18:
@HGSilveri you should be able to do it now.

HGSilveri, 2021-01-18:
Ok, the app is installed on this repo. If you want to use it, go [here](https://app.reviewnb.com/), sign up and you should be able to select Pulser. It's not of much use right now, but once we have a PR with changes to notebooks, we should be able to try this out.

sebgrijalva, 2021-01-18:
Thanks, this looks nice!

HGSilveri, 2021-01-18:
Does it work for you, @sebgrijalva ?

sebgrijalva, 2021-01-18:
Yes, I think so 

[image]

HGSilveri, 2021-01-18:
Ok, great!

closed 2021-01-18 by HGSilveri