pasqal-io/Pulser #67: Serializing a Sequence

HGSilveri, 2021-01-15:
In order to send a `Sequence` over to the QPU, there will need to be some sort of serialization. Although JSON is the obvious choice (mainly because it's the only one I know...), I'm sure there are other options, which might be better suited for the job. What's your take on this, @bejito ?

Another thing to decide here is whether or not we want to deserialize back to a Pulser Sequence. Currently, the translation from Pulser to the QM program is built to take in the `Sequence` object, so that would be a point in favour of having a mechanism for deserialization. Apart from the work involved (which I would risk saying is not tremendous), do you find any points against building this?

lhenriet, 2021-01-15:
I would be in favor of having a deserialization procedure. There is another usecase, in addition to the pulser -> QM translation that you mention. If a user chooses to 'emulate' the dynamics on Pasqal Cloud Services (and not locally), we will re-create the sequence to feed it into Simulation.

HGSilveri, 2021-01-15:
Ok, cool. I think that's enough to justify doing it. Now it's just a matter of deciding which format we'll choose (ie JSON or something else).

bejito, 2021-03-16:
Yes, agree with all the above.
In terms of languages, JSON is okay I think. If we feel like we need more perf in terms of bandwidth at some point, we can go with protobuf. Although more efficient, it is also less human readable so definitely not a good take for the dev phase. 
Serialization should be rather direct. There might be a little difficulty on the deserialization part with the variables since the sequence builder is build with the fact that a given variable's references are the same accross the builder. But using the fact that variables' names are unique accross a sequence builder should be enough to solve this.

closed 2021-04-13 by HGSilveri