pasqal-io/Pulser #68: Assess the effect of pulse duration restrictions on the tutorials

HGSilveri, 2021-01-15:
The restriction of a pulse duration to multiples of 4 ns might have unwanted effects on the results of two tutorials in particular: "Preparing an antiferromagnetic state in the Ising model" and "Tutorial MIS & QAOA".

I kindly ask the main authors, @sebgrijalva and @cdalyac, to verify whether the automatic rounding that is now occurring affects the expected outcomes and, if so, to adapt the notebooks accordingly.

sebgrijalva, 2021-01-15:
"Preparing state with antiferromagnetic order in the Ising model.ipynb" is running fine. I also ran "Tutorial MIS & QAOA.ipynb" and it doesn't seem to produce any unexpected output. Can you confirm @cdalyac ?

cdalyac, 2021-01-18:
"Tutorial MIS & QAOA" is working fine, warning still appears and results displayed are expected.

closed 2021-01-18 by cdalyac