pasqal-io/Pulser #74: Drawing the Rydberg radius

HGSilveri, 2021-01-21:
Just an idea that might be useful: we could add the option to draw the Rydberg blockade radius around the atoms in the Register, in typical unit-sphere fashion.

Further on, we could even combine this with a routine to extract the maximum Rabi frequency from a sequence, to automatically determine the Rydberg blockade radius. 

Another idea would be to take in the Rydberg radius and spit out the corresponding graph, like we do in MIS problems. Something like a `Register.to_graph(blockade_r)` method.

What do you think @lhenriet , @sebgrijalva , @cdalyac ?

cdalyac, 2021-01-21:
Good idea and I agree, it felt like this part was missing. 100% would use.

sebgrijalva, 2021-01-21:
Seems like a cool idea! 

> take in the Rydberg radius and spit out the corresponding graph, like we do in MIS problems. Something like a Register.to_graph(blockade_r) method.

I'm not sure I follow, how will you determine the corresponding graph? i.e. what is your initial information?

HGSilveri, 2021-01-21:
If two nodes are less than `Rb` away, they are connected by an edge.

HGSilveri, 2021-02-16:
Hey everyone! I'm implementing this thing, it's pretty much done. However, I have a question for you: how do you want to deal with edge cases where the atoms are exactly the Rydberg blockade radius away? In the example above, you see how rounding errors connect some atoms but not others, simple because of the rounding. 
[image]

This is the case we don't want, so I would either give a bit of margin above or below, so that every atom that's `Rb` away is either connected or disconnected. What's your take on this @sebgrijalva @cdalyac ?

cdalyac, 2021-02-16:
I'd say we give a bit of margin below so that if the effective distance is r_b + epsilon,  (with epsilon = `1e-8` as discussed with @HGSilveri ) we still see the link.

PR #87 (2021-02-18): Drawing blockade radius and connectivity graph
Here's the implementation of the drawing of the blockade radius and the corresponding connectivity graph. I implemented it in some of the tutorials, which I would ask you to review as well to see if they benefit from it.

As a bonus, I added the method `PasqalDevice.rabi_from_blockade`  to get the Rabi frequency that gives a certain Rydberg blockade radius because I felt it could be useful when the user has a given array and is trying to figure out their Sequence's parameters.

Closes #74 .
closed 2021-02-18 by HGSilveri