pasqal-io/Pulser #98: Special operators for waveforms

HGSilveri, 2021-03-05:
I think it would be a nice improvement to have some special operators for waveforms. I came up with these ones:
 - Negation: Doing `-wf` would return a Waveform with all it's samples' sign flipped
 - Multiplication: Doing `wf * 2` would multiply all it's values by 2. It doesn't make sense to multiply two waveforms.
 
 Not so sure these make sense:
 - Addition: Doing `wf + 2` would add 2 to all values. `wf1 + wf2` would concatenate two waveforms.
 - Absolute value
 - Power
 - Modulo

Feel free to discuss the pertinence of each or add more suggestions.

Louis-PaulHenry, 2021-03-16:
I'm just thinking about that now, but how about rescaling the time or the duration of a pulse, it might be a useful method : 
pulse too short -> increase the duration by time delta t, or multiply the timescale by a scale lambda (taking care of the 4 ns unit)?

HGSilveri, 2021-03-17:
I'm not sure I fully understand what you have in mind. I would say that rescaling is misleading when you are so constrained in the values you allow, as you will either force the user to carefully select the scaling factor or you'll have to change the outcome under the hood, which might give rise to undefined behaviour.

Louis-PaulHenry, 2021-03-17:
I'm just thinking of users who might try different pulse times to try and optimise their problem and try a given shape for different durations. It's easy enough to do by hand, but for a compositewaveform it might not be as straightforward.
I'll write a separate issue both for that and the rescaling of the area

PR #105 (2021-03-17): Implement multiplication by scalar and negation on waveforms
Implements negation, scalar multiplication and scalar division of waveforms. 

All operations change the waveform's samples, not the duration.

Closes #98 .
closed 2021-03-18 by HGSilveri