pasqal-io/Pulser #100: Effective Pulse Sequence in Simulations

sebgrijalva, 2021-03-09:
When a `sampling_rate` is specified in a `Simulation`,  a fraction of the samples is chosen to create the `QobjEvo` of the Hamiltonian. If the sampling rate value is very small, this may result in an oversimplified sequence. On the other hand, if one is looking for a good compromise of a low `sampling_rate` that runs faster and still faithful to the ideal sequence, having a visual aid is valuable. 

It would be useful to have a `Simulation.draw()` method (or maybe a better suited name) such that the user can see what  the resulting pulse sequence looks like.

All the information is stored in `Simulation.samples` so one solution would be to plot:

```python
def _draw_effective_seq(my_sim):
    samples_a = my_sim.samples['Global']['ground-rydberg']['amp']
    samples_d = my_sim.samples['Global']['ground-rydberg']['det']
    times = (1000*my_sim._times).astype(int)
    plt.figure(figsize=(14,6))
    plt.plot(times, samples_a[times], '.', ls='-')
    plt.plot(times, samples_d[times], '.', ls='-')
```

This is an example with only `Global` pulses in a 'ground-rydberg' basis, so a more robust method should look in all the provided samples dictionary. I also haven't included any phases that the pulses may have.

HGSilveri, 2021-03-09:
I would advise you to reuse the code in `_seq_drawer.py`. From glancing at the code there, it might be enough to adapt the `gather_data` function for when it receives a `Simulation` object instead of a `Sequence` one.

Louis-PaulHenry, 2021-03-15:
I've implemented a version of this in a new branch. 
Quick question about it : is there a reason why the "max_step" for sesolve is set to 5? Have you tried different values and settled on this one?

sebgrijalva, 2021-03-15:
The maximum step was set to 5 to avoid skipping parts of the sequence (the solver sometimes takes large jumps). Generally, it should be set as less of half the length of a given pulse.  I remember `max_step=5` being good enough for faster runtimes and no mistakes, but we didn't try tuning it further.

closed 2021-05-17 by HGSilveri