pasqal-io/Pulser #123: Strange behavior in the time series of the sequence adder

Louis-PaulHenry, 2021-04-12:
I've stumbled upon this issue while trying to get the 'spline extrapolated' pulse used by Qutip: 
when adding a pulse to the sequence, the resulting list of times `gather_data(seq)[ch]['time']` is not always strictly increasing, jumping back one time step at the junction between the two pulses (This might have to do with the leading 0 in the list).

HGSilveri, 2021-04-12:
Last time I checked, we weren't using `gather-data` for what is given to the Qutip solver. Are you doing this yourself?

Louis-PaulHenry, 2021-04-12:
I'm using `gather_data` myself to check. The issue was raised by `pytest` when trying to draw the sequence, and using the times from `gather_data`

HGSilveri, 2021-04-12:
So if the error comes from drawing, were does Qutip come in?

Louis-PaulHenry, 2021-04-12:
The Qutip does not really come in. It's just that Qutip is using spline interpolation and I'm trying to plot this interpolation (not necessarily via Qutip, currently by extracting the data 'by hand')

HGSilveri, 2021-04-12:
But you want to plot the interpolation that Qutip is actually doing in `Simulation`, or just an interpolation of the data used for plotting?

HGSilveri, 2021-04-12:
> edit : `gather_data` is just a shortcut to get the discretized pulse that Qutip is using

That's my point, the data Qutip is using does not come from `gather_data`

Louis-PaulHenry, 2021-04-12:
I want to plot the interpolation that Qutip is actually doing, but without having to wait for the simulation to be actually run. I'm not sure I can get Qutip to give me beforehands the extrapolated times. I can easily get the hamiltonian at intermediate times, but not the pulse, no?

HGSilveri, 2021-04-12:
I don't know how to do that with Qutip, or if Qutip is even useful. I would look into the interpolator that Qutip is using and use that instead. I can fix this error you're getting, but bear in mind that it only comes from an improper use of `gather_data`, which is not meant to provide data for simulation or interpolation.

I'll push a commit that should solve this issue.

Louis-PaulHenry, 2021-04-12:
That's what I tried to do. I just need to find another way than `gather_data` to extract the discretization of the puse.

Louis-PaulHenry, 2021-04-12:
`get_hamiltonian` works... for a system of a single atom, where one can easily read out $\delta$ and $\Omega$ from the Hamiltonian...
[image]

HGSilveri, 2021-04-12:
Cool, awesome!

Louis-PaulHenry, 2021-04-12:
I still need to do it for generic Hamiltonians/pulses, but the time ordering issue is solved, so I'll close this issue

closed 2021-04-12 by Louis-PaulHenry