pasqal-io/Pulser #147: Add Support for Simulation in XY Mode

sebgrijalva, 2021-05-11:
With the recently added support for "XY Mode" in a Pulser sequence (Refer to [This reference](https://arxiv.org/pdf/1408.1055.pdf) for some background on the physics behind), it remains to add the corresponding simulation. This implies adding an XY Hamiltonian which takes as time-dependent coefficients the pulses of the sequence, and will imply several changes to the `Simulation` class:

- A  `self._interaction` or self._sequence_mode` private attribute will allow to classify the incoming sequence depending on the chosen mode.
- The `self.samples` dictionary has to correspond to the case of 'Ising' or 'XY' interaction: In the case of the latter, since there is only one basis (`'xy'`), we only need slots for `'Global'` and `'Local'`.
- Since the sequence has been verified to have consistent basis of pulses throughout, the `write_samples()` function inside the `self._extract_samples()`method would only be slightly modified. On the other hand, the `self._build_basis_and_op_matrices()` method should now reflect the two-Rydberg level nature of the interaction , while following the existing scheme of the (actually more complex) Ising interaction. Perhaps just call these levels 'up' and 'down'?
- In the `make_interaction_term()` function of the `self._construct_hamiltonian()` method, one needs to add the proper term `0.5 * C_3 / R_ij**3`, which depends on the cube of the interatomic distance and has now a different coupling constant `C_3` (which should be taken as  `C_3 = 3.7 10**3 Ghz * µm**3`. 
- In fact, the coefficient `C_3` depends on the relative position of the atoms (Cf. Supplementary Material of [the Reference](https://arxiv.org/pdf/1408.1055.pdf)) as `C_3(theta) = C_3 (1 - 3cos(theta)**2)`. It's already good if one can introduce (to begin with) a constant coefficient `C_3`, but If one could read from the register the position of the corresponding atoms and construct the more realistic expression, it would be a very welcome addition.
- The `build_coeffs_ops()`function inside `self._construct_hamiltonian()`should also take into account the differently built operators.
- To check things are working correctly, one could try to add a test displaying Rabi Oscillations with similar properties as that of Fig. 1(c) of the [Reference](https://arxiv.org/pdf/1408.1055.pdf), with one and then several atoms in an array and global/local pulses.

HGSilveri, 2021-05-11:
Closing this just until #144 is merged.

Slimane33, 2021-05-24:
Hello,
I would like to do it.

HGSilveri, 2021-08-03:
**Status update:**

Support for XY Simulation has been successfully added and is currently residing in the `xy-simulation` branch. Since the work on this feature started, the `develop` branch has diverged to incorporate support for noisy simulations. Therefore, the changes in `xy-simulation` now need to be reconciled with those in the `develop` branch, with the goal of ultimately merging `xy-simulation` into `develop`.

HGSilveri, 2021-08-10:
**Status update**:

The `xy-simulation` has been (apparently) successfully merged into `noisy-xy` (which was branched from `develop`). This issue can then be closed once `noisy-xy` is merged into `develop`.

PR #264 (2021-09-21): Adding support for XY Simulation
- Adds simulation capabilities in the XY mode, as described in detail #147 and originally executed in #198 
- Combines the changes of #198 with the latest version of `pulser.simulation`, which supports noisy simulation. In this PR, only support for simulation of `"SPAM"` noise in the XY mode is added.
- Adds an SLM Mask feature for initial state preparation (as described in #248 and executed in #258)

Closes #147.
closed 2021-09-21 by HGSilveri