pasqal-io/Pulser #150: Use static channel names

romainfd, 2021-05-11:
Could `seq.declare_channel('ch0', 'raman_local')` be replaced by something like this
```
from pulser import Channels
seq.declare_channel('ch0', Channels.RAMAN_LOCAL)
```
(or using `Constants` instead of `Channels`)

I think this would make the project a bit more robust and also easier to use as autocomplete in IDEs would pick these up.

HGSilveri, 2021-05-11:
This one is not so simple, because the channel's specs change between devices. When you choose the `raman_local` channel, you are effectively saying you want the channel named `raman_local` of the device chosen for this `Sequence` (in this example, `Chadoq2`). So, in fact, there can be multiple `raman_local` channels with different characteristics. 

The way it is currently done, you're supposed to not need to look into the device because of the `Sequence.available_channels` property, which fetches and filters the devices available for declaration. 

I admit I'm not a big fan of the solution we came up with, particularly when it comes to using strings to identify the channels, but I still think it's the best one so far.

HGSilveri, 2021-05-19:
Closing this issue, as the channels can't be static as suggested.

closed 2021-05-19 by HGSilveri