pasqal-io/Pulser #153: Cubic spline extrapolation of the Hamiltonian by QuTip

Louis-PaulHenry, 2021-05-12:
When building the Hamiltonian QobjEvo, the entire list of samples of a Sequence is given to QuTip. 
QuTip will then extrapolate (with Cubic spline) the values of the coefficients in between. This leads to artifacts when defining a piecewise constant pulse, at the transitions between plateaus. 
One way to correct that would be to split the Sequence into constant and time-dependant parts and tell QuTip about it.
From QuTip documentation :

> The times of the tlist do not need to be equidistant, but must be sorted. By default, a cubic spline interpolation will be used for the coefficient at time t. If the coefficients are to be treated as step functions, use the arguments `args = {"_step_func_coeff": True}`.

Louis-PaulHenry, 2021-08-16:
Now that I'm thinking about it. Another way to solve this issue would be to discard the sampling times altogether and to build the Hamiltonian QobjEvo by giving the functional dependence of the coefficients. This way Qutip doesn't do any interpolating

Louis-PaulHenry, 2021-09-08:
Using the `_step_func_coeff` argument in `sesolve` only gives a ~20% speedup when the Pulse is *completely* piecewise constant. Otherwise, it only gives a small increase in accuracy, at the expanse of a large slowing down of the computation.

Here is a short notebook illustrating the effect :
[pwc.ipynb.zip](https://github.com/pasqal-io/Pulser/files/7127711/pwc.ipynb.zip)

aarontrowbridge, 2026-10-23:
A data point that this is still present, and can matter much more than the earlier notebook suggested: on pulser 1.9.1 / pulser-simulation 1.9.1 / qutip 5.3.1, a 4 ns piecewise-constant detuning (`CustomWaveform`, ±10 rad/µs) is integrated as a cubic spline between samples. Within a single 1 ns sample it swings through zero and overshoots:

```python
SNIPPET
```

```
OUTPUT
```

For smooth pulses this is invisible (we see ~1e-6 agreement). For optimized, jagged piecewise-constant pulses it isn't: on a two-atom Rydberg CZ pulse with 4 ns detuning steps, the gate fidelity from `QutipEmulator` differed from a piecewise-constant reference by **0.125**. Rebuilding the same Hamiltonian with qutip's `coefficient(..., order=0)` and `sesolve` brought it back to 1e-6 agreement with an independent matrix-exponential rollout.

qutip 5 makes the fix a small one: `qutip.coefficient(array, tlist=times, order=0)` gives a step-function coefficient (the qutip-4 `_step_func_coeff` arg is gone). A user-facing option such as `QutipEmulator.from_sequence(..., interpolation="cubic" | "linear" | "constant")` (or an entry on `QutipConfig`), defaulting to the current behaviour, would let piecewise-constant users opt in without changing anyone else's results.