pasqal-io/Pulser #177: Global Rydberg pulse is only single threaded

paoloviviani, 2021-05-25:
I tried to run execute the [UD-MIS tutorial](https://github.com/pasqal-io/Pulser/blob/develop/tutorials/applications/Using%20QAOA%20to%20solve%20a%20MIS%20problem.ipynb) either on the proposed register and on a larger one (10 quits) with longer sequences/more layers.
With typical pulser/qutip installation from pip the single quantum loop runs always on a single core only.

I tried to:

- compile qutip from source against OpenMP
- use qutip installed with Conda and linked agains MKL
- modify the [Pulser simulation parameters](https://github.com/pasqal-io/Pulser/blob/4e5213460982492730fd4e3cd2b85fa6cae31c56/pulser/simulation/simulation.py#L337) to pass explicitly [multi-threading options to qutip solver](https://github.com/pasqal-io/Pulser/blob/4e5213460982492730fd4e3cd2b85fa6cae31c56/pulser/simulation/simulation.py#L337) (num_cpus, use_openmp, openmp_threads).

but none of the solutions above improved the cpu usage beyond a single core. 

In general it would be possible to parallelise the code at a different level (e.g. at the level of the optimization loop, like with population-based optimizers), however, I was wondering if this is the expected behaviour or if there is a chance of parallelising the simulation itself.

sebgrijalva, 2021-05-25:
Hi Paolo, 

Thanks for raising this issue, indeed, It's interesting to try to run a multi-thread job for Pulser leveraging QuTiP. I will try to reproduce a basic approach and get back to you. We might have to ask the devs at QuTiP too if necessary.

sebgrijalva, 2021-05-27:
Here's a simple implementation of using multiple cores with Pulser and leveraging QuTiP's `parallel_map()` function:

```python
import numpy as np
import qutip

from pulser import Register, Pulse, Sequence
from pulser.devices import Chadoq2
from pulser.simulation import Simulation

# Create a parametrized sequence
reg = Register.from_coordinates([[0,0]] )  # Just one atom.
seq = Sequence(reg, Chadoq2)
seq.declare_channel('ch0', 'rydberg_global')

rabi = seq.declare_variable('rabi')

seq.add(Pulse.ConstantPulse(200, rabi ,0, 0), 'ch0')
```

Notice that instead of assigning values one by one in a `for` loop (and thus having to redefine several times the `Sequence`instance), I am using a parametrized sequence with variable `rabi`. Then, I will build each sequence according to a given `param`:

```python
# Define simulation experiment function:
def exp(param):
    obs = qutip.basis(2,0).proj()  # Calculate occupation in the rydberg state
    seq_instance = seq.build(rabi=param)
    seq_instance.measure('ground-rydberg')
    
    sim = Simulation(seq_instance, sampling_rate=0.01)
    res = sim.run()

    return res.expect([obs])[0]
```

With this setup you can use QuTiP's `parallel_map()` function which automatically uses all the cores available in your computer :

```python
qutip.parallel_map(exp, np.linspace(1, 1.5, 100)) # Get expectation values for the entire dynamics and 100 parameter values
```

Notice this will be twice as fast as simply using

```python
for val in np.linspace(1,1.5,1000):
    exp(val)
```

Depending on the parameters of your sequence, this might be what you're looking for. Let me know if it helps.

HGSilveri, 2022-05-10:
@sebgrijalva This issue has been open for a while now, what's the status on this? Should we close it / archive it?

sebgrijalva, 2022-05-10:
I'd say we archive it, yes.