pasqal-io/Pulser #181: Potential conflict between `pulser.Sequence` and `collections.abc.Sequence`

HGSilveri, 2021-05-27:
From adding type hints, I realized that there are built-in objects called `Sequence`, namely those of `collections.abc` and `typing`. This can create conflicts if we're not careful. 

In case we want to avoid this, we will have to consider renaming the `pulser.Sequence` class and deprecating the `Sequence` name.

LaurentAjdnik, 2021-05-31:
What a pity, `Sequence` was such a cool and meaningful name! :grin:

Would the consequences be that bad?

Do you have any clue on how to rename it?

My 20 cents: `Timeline`? `Flow`? `Progression`? `Procession`? `Run`? `Flow`? `Chronology`? `Arrangement`? `Succession`? `Chain`? `Train`?...

Or by concatening `Sequence` with something else? `PulseSequence` ? `Sequencer`?

HGSilveri, 2021-05-31:
> Would the consequences be that bad?

A bit, considering the central role `Sequence` plays. Virtually all existing user scripts would stop working or raise a `DeprecationWarning` if such a change were to take place. 

> Do you have any clue on how to rename it?

I was thinking along the lines of a concatenation, yes, something like `PulseSequence` or `PulserSequence`. However, changing the name still sounds a bit drastic to me, specially because this conflict only needs to be taken into account by the developers, not the users.

closed 2021-06-28 by HGSilveri