pasqal-io/Pulser #187: Add a way to export figures from `Sequence.draw`

Slimane33, 2021-06-02:
I am thinking of adding a feature to the `Sequence.draw()` method in order to be able to export the generated figure like `plt.savefig()`.

sebgrijalva, 2021-06-02:
Looks like an interesting option. I personally tweaked the code when adding pictures for the ArXiv note in order to change size, resolution or output type; it was basically a matter setting kwargs in the `plt.plot()` method (which is hidden from the user at the time). Maybe adding that choice in `_seq_drawer` could be a way to proceed. 

In fact, the same feature could be added potentially to all plotting methods. What do you think?

HGSilveri, 2021-06-02:
> Looks like an interesting option. I personally tweaked the code when adding pictures for the ArXiv note in order to change size, resolution or output type; it was basically a matter setting kwargs [...]

I think we should consider carefully what parameters and options are actually useful. I think that the ability to save the figure to a file is useful, and perhaps we should also give access to some of the relevant parameters of `plt.savefig()`, but probably not all indiscriminately.

Slimane33, 2021-06-02:
What if we suppress the `plt.show()` at the end of the `_seq_drawer` and return the current `Axes` ? In this way it is an object that the user can later manipulate.

HGSilveri, 2021-06-02:
Yes, but it will break the existing usage. I say make `_seq_drawer` return the `Axes`, and then have a method that `draws` it (`Sequence.draw`). and one that saves the plot (`Sequence.savefig`)

Slimane33, 2021-10-05:
It could be also nice to export the register figures

closed 2021-11-09 by HGSilveri