pasqal-io/Pulser #189: Adding a `.gitattributes` file

paniash, 2021-06-03:
Hi!

Even though `Pulser` is a python package, on first glance it seems that a large portion of the software is `.ipynb` files as opposed to `.py` files. This gives a false impression of the package being written in jupyter notebooks.

Not that it matters, I'm just being pedantic but would be nice to consider adding a `.gitattributes` file.

paniash, 2021-06-03:
If one agrees, I'll be more than happy to open a PR for the same.

HGSilveri, 2021-06-04:
> Not that it matters, I'm just being pedantic but would be nice to consider adding a `.gitattributes` file.

Not at all, I agree. Thanks for the suggestion, feel free to proceed with the PR.

paniash, 2021-06-04:
Fixed in [` 698d1c6 `](https://github.com/pasqal-io/Pulser/commit/cfeda9d3f00f3ee358e5b27a4c4ca2e201345a0e).

closed 2021-06-04 by paniash