pasqal-io/Pulser #203: Add __getitem__ to Waveform

HGSilveri, 2021-06-14:
The idea would be to have an easy access to the samples of a Waveform, preferentially without computing the full array of samples. Currently, the properties `first_value` and `last_value` do this for the first and last values, but a `__getitem__` would replace them and extend the functionality to the entire waveform. 

The keys would be the time in `ns`, also allowing specification from the end (e.g. `t=-1`) and slicing.

LaurentAjdnik, 2021-06-16:
You can assign me.

HGSilveri, 2021-06-16:
@LaurentAjdnik Great! There's a bit of nuance here, but I'm sure you're up to the task. A few pointers:
- We want just `__getitem__`, not `__setitem__`. By design, a waveform should not be changed in-place after initialization
- A big part of the motivation to do this is to save in computation time and/or memory by avoiding computing `samples` or, if it makes sense to compute it, do it just once. If we didn't care about this, we would just get items from `Waveform.samples`. You will see in the `first_value` and `last_value` examples on how this can be done. In the cases where it makes sense to compute the `samples`, it's worth considering some sort of caching mechanism.

Thanks again, and feel free to reach out with any questions!

LaurentAjdnik, 2021-06-16:
Lots of questions raised indeed. I will draft-PR something soon and we can discuss based on that.