pasqal-io/Pulser #208: Change default value in `max_step` for simulations.

sebgrijalva, 2021-06-18:
The current default value in `max_step` is currently set to 5, which was originally thought as nanoseconds. However, the simulations are now given in microseconds. It should be set to a smaller value or altogether eliminated. 

If we eliminate it, the user would now have to be responsible for setting up correct simulation parameters. This hasn't been an issue with the current values for most simulations (the value of 5 would correspond to 5000 ns), but this is not the first time some unwanted bugs appear (see #207 )

HGSilveri, 2021-06-18:
In the cases where the user does not specify it, I would suggest some sort of sensible criteria. For example, we could check if the Hamiltonian starts off diagonal and, in those cases, set `max_step` to a value to avoids the behaviour of #207, instead of relying on the solver to decide the value.

sebgrijalva, 2021-06-18:
Another option is to check all (or keep track of) the duration of delays and pulses and set `max_step` to half the minimum value. However I haven't checked about any cost a small maximum step would have in run time.

closed 2021-07-08 by sebgrijalva