pasqal-io/Pulser #228: Implement Kaiser waveform

HGSilveri, 2021-07-01:
### Discussed in https://github.com/pasqal-io/Pulser/discussions/227

<div type='discussions-op-text'>

<sup>Originally posted by **LaurentAjdnik** July  1, 2021</sup>
Hi all!

From https://arxiv.org/abs/2104.15044:

`In practice, a properly shaped waveform—like the Blackman waveform, which we will use here — can mitigate unwanted modifications to the modulated signal, stemming from spectral leakage or noise sensitivity.`

From https://numpy.org/doc/stable/reference/generated/numpy.blackman.html:

`The Blackman window is a taper formed by using the first three terms of a summation of cosines. It was designed to have close to the minimal leakage possible. It is close to optimal, only slightly worse than a Kaiser window.`

Hence, should we implement a `KaiserWaveform` based on https://numpy.org/doc/stable/reference/generated/numpy.kaiser.html (probably with `beta = 8.6`)?
</div>

To implement, with `beta=14` by default, as suggested in numpy's documentation.

HGSilveri, 2021-07-01:
@LaurentAjdnik please comment on this so that I can assign it to you

LaurentAjdnik, 2021-07-01:
> @LaurentAjdnik please comment on this so that I can assign it to you

Me, me, me. 😆

PR #229 (2021-07-13): Add KaiserWaveform class (#228)
Adds KaiserWaveform class, based on `numpy.kaiser()`.

Adds the corresponding tests.

Fixes #228.
closed 2021-07-13 by HGSilveri