pasqal-io/Pulser #249: Classical simulation does not allow simulation time equal to the duration of the pulse

Adrien-Kahn, 2021-08-16:
When simulating a 4ns sequence, `res.states` only contains 4 states, including the initial state. However, since 4 values for the Hamiltonian were provided, the initial state could be updated 4 times, so that we would have a total of 5 states in `res.states` with the initial state.

Manually setting the `evaluation_times` of the `simulation` does not help because the setter checks that the largest time is smaller than `self.times[-1]` which is equal to the duration of the sequence minus 1.

Adrien-Kahn, 2021-08-17:
Here is a minimal working example:
```
reg = Register.rectangle(1,2)
p = Pulse.ConstantPulse(4, 40, 80, 0)
seq = Sequence(reg, devices.MockDevice)
seq.declare_channel('ryd','rydberg_local', 0)
seq.add(p1, 'ryd')

simtest = Simulation(seqtest)
restest = simtest.run()

# Overlap with |gg>
print([abs(state.overlap(q.basis(4,3)))**2 for state in restest.states])
```

Output:
```
[1.0, 0.9996002665950946, 0.998404263138335, 0.9964215525372951]
```

The first value corresponds to the initial state, so we are only given 3 updated states, while 4 values for the Hamiltonian were provided.

sebgrijalva, 2021-10-06:
While implementing these changes, there appeared some issues related to the interpolation of the call method of the `QobjEvo`. I'm still working on this, but it will take longer than expected.

closed 2021-11-25 by HGSilveri