pasqal-io/Pulser #250: Global Ising with noise and local Ising channels might overwrite on each other

darcangelomauro, 2021-08-17:
A global pulse with any noise (but dephasing) is translated into a local pulse on each qubit. 

The way the code is written now, if a local Ising pulse and a global Ising pulse with noise act on the same qubit at the same time, the two pulses do not add, but rather the last one that is processed overwrites the other.

Attached you can find a minimal working example.

[noise_bug.ipynb.zip](https://github.com/pasqal-io/Pulser/files/6998816/noise_bug.ipynb.zip)

sebgrijalva, 2021-08-17:
Here's the minimal code explicit for future reference and testing:

```python
from pulser import Pulse, Sequence, Register
from pulser.simulation import Simulation, SimConfig
from pulser.devices import Chadoq2

qubits = {"q0":(0,0)}
reg = Register(qubits)

seq = Sequence(reg, Chadoq2)

seq.declare_channel('ch_local', 'rydberg_local', initial_target='q0')
seq.declare_channel('ch_global', 'rydberg_global')

pulse = Pulse.ConstantPulse(20, 10, 0, 0)

seq.add(pulse, 'ch_local')
seq.add(pulse, 'ch_global', protocol = 'no-delay')

# No noise added: samples are correctly split into 
# global and local, giving a total amplitude of 10 (recall factor of 2 in Hamiltonian)

sim_no_noise = Simulation(seq)

print("No noise:")
print(sim_no_noise.get_hamiltonian(5),"\n")

# With noise (not dephasing): global is translated into local
# and the two will overwrite each other, resulting in an amplitude of 5 instead of 10

sim_with_noise = Simulation(seq)
config_doppler = SimConfig(noise = ('doppler'))
sim_with_noise.set_config(config_doppler)

print("Noise added:")
print(sim_with_noise.get_hamiltonian(5)

sim_no_noise.get_hamiltonian(5) == sim_with_noise.get_hamiltonian(5)
```

One should get `True` from the final assertion.

sebgrijalva, 2021-08-17:
This one seems fit for a hotfix into the `master` branch. I will create the corresponding branch. To be sure I'm not skipping any places where this is corrected, would you like to fix the bug, @julien-bremont ?

darcangelomauro, 2021-08-18:
I might be wrong, but the fix could be as simple as replacing `=` with `+=`  in the following block:
```          
            samples_dict["amp"][slot.ti : slot.tf] = (
                _pulse.amplitude.samples * noise_amp
            ) 
            samples_dict["det"][slot.ti : slot.tf] = (
                _pulse.detuning.samples + noise_det
            )
            samples_dict["phase"][slot.ti : slot.tf] = _pulse.phase
```
in `Simulation.write_samples`

darcangelomauro, 2021-08-23:
I believe in @sebgrijalva's minimal example one should compare the off-diagonal components of the Hamiltonian, not the whole thing, because doppler noise adds a random detuning term.

So something like this:

```python
from pulser import Pulse, Sequence, Register
from pulser.simulation import Simulation, SimConfig
from pulser.devices import Chadoq2

qubits = {"q0":(0,0)}
reg = Register(qubits)

seq = Sequence(reg, Chadoq2)

seq.declare_channel('ch_local', 'rydberg_local', initial_target='q0')
seq.declare_channel('ch_global', 'rydberg_global')

pulse = Pulse.ConstantPulse(20, 10, 0, 0)

seq.add(pulse, 'ch_local')
seq.add(pulse, 'ch_global', protocol = 'no-delay')

# No noise added: samples are correctly split into 
# global and local, giving a total amplitude of 10 (recall factor of 2 in Hamiltonian)

sim_no_noise = Simulation(seq)

print("No noise:")
print(sim_no_noise.get_hamiltonian(5),"\n")

# With noise (not dephasing): global is translated into local
# and the two will overwrite each other, resulting in an amplitude of 5 instead of 10

sim_with_noise = Simulation(seq)
config_doppler = SimConfig(noise = ('doppler'))
sim_with_noise.set_config(config_doppler)

print("Noise added:")
print(sim_with_noise.get_hamiltonian(5))

sim_no_noise.get_hamiltonian(5)[0,1] == sim_with_noise.get_hamiltonian(5)[0,1]
```

where the last statement should return `True`

darcangelomauro, 2021-08-25:
I added a pull request for the issue directly into `master`, correct @sebgrijalva?

sebgrijalva, 2021-08-25:
Yes, hotfixes are meant to be only from and back to the `master` branch. Unfortunately you merged `develop` to the hotfix branch and this should be avoided until we are ready to merge many new changes to `master`. 

But since its such a minor difference this time, I can edit the difference.

sebgrijalva, 2021-08-25:
Closed by #254

closed 2021-08-25 by sebgrijalva