pasqal-io/Pulser #257: Implementation of `Register.max_connectivity` with tabulated circle packings

Adrien-Kahn, 2021-08-30:
Currently, `Register.max_connectivity` is implemented with a triangular lattice which is not optimal, in particular for small number of atoms.
The problem of creating a register with maximum connectivity (more precisely, to minimize the maximum distance between two atoms) can be reformulated as the [circle packing problem](https://en.wikipedia.org/wiki/Circle_packing_in_a_circle), where one tries to find the way to pack a given number of unit disks within the smallest possible circle. The simplest example where this approach performs better is for 4 atoms, where the triangular lattice gives a diamond, while the optimal circle packing would give a square. See for example the best known packing of 10 disks and how it compares with the triangular lattice:

[image]
[image]

The best known circle packings are tabulated for a number of disks up to 2600: http://hydra.nat.uni-magdeburg.de/packing/cci/ so such an implementation would only have to fetch the tabulated coordinate.

sebgrijalva, 2021-08-30:
Hi @Adrien-Kahn , this is an interesting topic of itself since the array geometry has consequences on many properties of the phase diagram of the ground-state as well as the "resource" hamiltonian of quantum optimization problems. In the N=4 case you describe, the square packing gives 4 "connections" or "edges", whereas the triangular lattice gives 5. `max_connectivity` (as discussed in #132 ) tries to maximize the number of edges or blockades between atoms, so the triagular geometry seems fit (they even went as far as including as much -triangular/hexagonal- symmetries as possible).

However, I think it's worth a try to implement some sort of `max_packing` analogous function, which as you say fetches from existing results. What would be some properties of the resulting connectivity graphs ? I can imagine that for someone trying to "chain" some transport properties in the array, these graphs may be attractive ;-)

HGSilveri, 2021-09-15:
Let's close this one for now.

closed 2021-09-15 by HGSilveri