pasqal-io/Pulser #268: Change/choice of n  / Customisation of C_6

Louis-PaulHenry, 2021-09-29:
The value of the interaction constant C6 is currently arbitrarily set. It might be useful to allow for the choice of a difference Rydberg level, and therefore a different value of C6 (as well as potentially introducing an anisotropic interaction in the case of D and P states)

LucasGitQ, 2021-10-05:
One way to do it without importing the whole ARC package (used to compute exactly C6) would be to precompute C6 for n=50,...,100 for instance and add a file with all those values in the **devices** folder. Then interaction_coeff = C6_list[rydberg_level]. Then a way to change an attribute of a frozen class is needed (see https://github.com/pasqal-io/Pulser/issues/263#issue-991155712). With this, a rydberg_level attribute could be changed (within a working bound) and the interaction_coeff updated at the same time.

LucasGitQ, 2021-11-16:
#281

closed 2021-11-22 by HGSilveri