pasqal-io/Pulser #284: Reorganize and Review Tutorials

HGSilveri, 2021-11-25:
- [ ] Reorganize the tutorials into more specific folders, `applications` is too vague at this point
- [ ] Review that they are all running well after all changes to `develop` are concluded

sebgrijalva, 2021-11-25:
I can help with reviewing the tutorials, specially some of the older ones.

HGSilveri, 2021-11-25:
> I can help with reviewing the tutorials, specially some of the older ones.

Thanks, that would be helpful, specially since the `interaction_coeff` increased slightly and I'm not sure of the effect that will have. At the least, I've been seeing some `ODEError`s.

I don't know what the best way to do this will be though. There are some tutorials that are intentionally left without output, so we should check their output explicitly in the built docs. The ones that take longer have the output saved, so we need to rerun them and commit the new outputs. For those, NBReview will do I guess

closed 2021-12-02 by HGSilveri