pasqal-io/Pulser #310: Type hints for (many) decorated functions

lvignoli, 2022-01-25:
Many methods are not hinted correctly by vscode [Pylance](https://marketplace.visualstudio.com/items?itemName=ms-python.vscode-pylance) language server.

Here are some example among others:

[image]
[image]
[image]

Even though the docstring is available, it makes the type hinting pointless in terms of user experience, with a bad "discoverability" of these functions usage. In particular, completion and suggestions are unavailable for the arguments.

It seems the decorators of these functions overwrite the type hints. For example, `Sequence.add()` receives the `@_store` decorator

https://github.com/pasqal-io/Pulser/blob/7405d8cd7463782891cbbf40335f163dc8f284cc/pulser/sequence.py#L89-L97

Removing all the type hints for the `_store()` and `wrapper()` functions restore the expected types, and Pylance behaves as expected:

```diff
diff --git a/pulser/sequence.py b/pulser/sequence.py
index aacbb6a..7289745 100644
--- a/pulser/sequence.py
+++ b/pulser/sequence.py
@@ -86,11 +86,11 @@ def _screen(func: Callable) -> Callable:
     return wrapper
 
 
-def _store(func: Callable) -> Callable:
+def _store(func):
     """Stores any Sequence building call for deferred execution."""
 
     @wraps(func)
-    def wrapper(self: Sequence, *args: Any, **kwargs: Any) -> Any:
+    def wrapper(self: Sequence, *args, **kwargs):
         def verify_variable(x: Any) -> None:
             if isinstance(x, Parametrized):
                 # If not already, the sequence becomes parametrized

```

Following [PEP 612](https://www.python.org/dev/peps/pep-0612/), I would suggest redefining decorators using the following pattern:

```python
from functools import wraps
from typing import Callable, ParamSpec, TypeVar

P = ParamSpec("P")
R = TypeVar("R")  # Return type

def counter(func: Callable[P, R]) -> Callable[P, R]:
    """Decorator keeping track of the number of calls of func."""
    count: int = 0

    @wraps(func)
    def wrapper(*args: P.args, **kwargs: P.kwargs):
        nonlocal count
        count += 1
        print(f"Call {count} of function {func.__name__!r}")
        return func(*args, **kwargs)

    return wrapper
```

But these features are 3.10 only, while we fix the type_hints to version 3.7 in `requirements.txt`.

If keeping the 3.7 requirement is necessary, the workaround would then simply be eliminating all type hints from the concerned decorators.
@HGSilveri @sebgrijalva what are your thoughts ?

HGSilveri, 2022-01-25:
@lvignoli Did you check if `mypy` complains if you remove the types from the decorators?

> But these features are 3.10 only, while we fix the type_hints to version 3.7 in requirements.txt.

Do you mean this? 
```
typing-extensions; python_version == '3.7'
backports.cached-property; python_version == '3.7'
```

It's just installing these two packages only when the version is 3.7, because they are built-in from 3.8 on. However, I would say that including version 3.10 only changes is a bit premature, so I would go with removing the type hints from the decorators if possible.

lvignoli, 2022-01-25:
@HGSilveri I have checked `mypy` and it does complain of the lack of types...

## Naive fix

Remove all type annotations and add `type: ignore` to prevent `mypy` complaining.

- it restores the correct hinting and Pylance suggestions
- `mypy` do not complain, as it does not check anything...

## Better fix

1. `*args` and `**kwargs` may be typed using `str` and `int` respectively as specified in [PEP 484](https://www.python.org/dev/peps/pep-0484/#arbitrary-argument-lists-and-default-argument-values).
2. [`mypy` docs](https://mypy.readthedocs.io/en/stable/generics.html#declaring-decorators) suggests how to type decorators using an explicit cast on the return function.

This gives a way to correctly type the decorators using `python>3.5`, if I am not mistaken.
Here an example on `@parametrize` from `pulser/parametrized/decorators.py`:

```python

F = TypeVar("F", bound=Callable[..., Any])

def parametrize(func: F) -> F:
    """Makes a function support parametrized arguments.

    Note:
        Designed for use in class methods. Usage in instance or static methods
        is not supported, and in regular functions is not tested.
    """

    @wraps(func)
    def wrapper(*args, **kwargs):
        for x in chain(args, kwargs.values()):
            if isinstance(x, Parametrized):
                return ParamObj(func, *args, **kwargs)
        return func(*args, **kwargs)

    return cast(F, wrapper)
```

 - it restores the correct hinting and Pylance suggestions
 - however, the correct types of `Pulse.ConstantDetuning()` now flow along the code. Hence, `mypy` complains about bad parameter types in some tests:
 
```log
pulser/tests/test_parametrized.py:31: error: Argument 2 to "ConstantDetuning" of "Pulse" has incompatible type "_VariableItem"; expected "float"
pulser/tests/test_parametrized.py:31: error: Argument 3 to "ConstantDetuning" of "Pulse" has incompatible type "_VariableItem"; expected "float"
Found 2 errors in 1 files (checked 39 source files)
```

It seems to me that _the current very permissive typing of these decorators functions has shadowed large part of the code base to `mypy` checks_.
Using either the above typing or the 3.10 one would likely highlight several type abuses, so I think it is an ungrateful but necessary task.

I attach the git diff of this change against current head of `develop` (commit `7405d8c`) for reproducibility:

```diff
diff --git a/pulser/parametrized/decorators.py b/pulser/parametrized/decorators.py
index 273e2a7..3da9774 100644
--- a/pulser/parametrized/decorators.py
+++ b/pulser/parametrized/decorators.py
@@ -18,12 +18,15 @@ from __future__ import annotations
 from collections.abc import Callable
 from functools import wraps
 from itertools import chain
-from typing import Any
+from typing import Any, TypeVar, cast
 
 from pulser.parametrized import Parametrized, ParamObj
 
 
-def parametrize(func: Callable) -> Callable:
+F = TypeVar("F", bound=Callable[..., Any])
+
+
+def parametrize(func: F) -> F:
     """Makes a function support parametrized arguments.
 
     Note:
@@ -32,10 +35,10 @@ def parametrize(func: Callable) -> Callable:
     """
 
     @wraps(func)
-    def wrapper(*args: Any, **kwargs: Any) -> Any:
+    def wrapper(*args: str, **kwargs: int) -> Any:
         for x in chain(args, kwargs.values()):
             if isinstance(x, Parametrized):
                 return ParamObj(func, *args, **kwargs)
         return func(*args, **kwargs)
 
-    return wrapper
+    return cast(F, wrapper)
```

HGSilveri, 2022-01-25:
> 1. `*args` and `**kwargs` may be typed using `str` and `int` respectively as specified in [PEP 484](https://www.python.org/dev/peps/pep-0484/#arbitrary-argument-lists-and-default-argument-values).

I think this is given as an example to show that you can type `*args` and `**kwargs`, but it's only valid if they are always the types you specify. 

> It seems to me that _the current very permissive typing of these decorators functions has shadowed large part of the code base to `mypy` checks_.
> Using either the above typing or the 3.10 one would likely highlight several type abuses, so I think it is an ungrateful but necessary task.

I agree, if we do find type abuses we should fix them. Can you please make sure that what you were getting was not because of the type hints for `*args` and `**kwargs`?

lvignoli, 2022-01-25:
> I think this is given as an example to show that you can type *args and **kwargs, but it's only valid if they are always the types you specify.

Indeed, I read too fast and thought it was a stub for unspecified types, but Any is already there, so my mistake...

> I agree, if we do find type abuses we should fix them. Can you please make sure that what you were getting was not because of the type hints for *args and **kwargs?

Putting `Any` on `*args` and `**kwargs` do not change `mypy` complains about `test_parametrized.py`. It seems the cast to `Callables[..., Any]`of the wrapper — along with the `Any` on the arguments — is enough to make the correct types flow in the code.

You may checkout [this branch](https://github.com/lvignoli/Pulser/tree/decorators-type-hints) to reproduce the failing checks.

closed 2022-01-26 by HGSilveri