pasqal-io/Pulser #315: Add option to simulate a Sequence with the modulated output

HGSilveri, 2022-02-02:
#312 introduces output modulation into Pulser. The next logical step is to use the expected output to get more reliable results out of a simulation. Important notes:

- The option to choose between using the input or the output samples should be given upon initialization of `Simulation`. For consistency with the current behaviour, the default should be to use the input.
- In principle, this change should only affect the samples that are extracted and used for simulation.
- Considering pulses in isolation can lead to mistakes, because `modulation(pulse1 + pulse2) != modulation(pulse1) + modulation(pulse2)`. 
- The correct approach is to consider the full channel output and modulate that, as is done when drawing the sequence. This is fine for global channels, but can be tricky to do for local channels because the samples are separated according to the target qubit. Note, however, that a sequence of pulses in between two target operations can be safely considered in isolation because the necessary fall time is included.

lvignoli, 2022-03-28:
Modulation is now handled at the sampler level with the merging of #345.
Issue #349 tracks the progress of using the sampler in simulations.

HGSilveri, 2022-07-07:
Something to take into account here: the behaviour of the SLM mask in the presence of modulation is still ill-defined, so it was decided that for now we will just not support the simulation of sequences with modulated output and an SLM mask. 

Practically, this implies that in `Simulation.__init__()`, if the user chooses the option to use the modulated output, we check whether the SLM mask is configured and raise a `NotImplementedError` if so.

PR #390 (2022-08-05): Simulation with modulated output
A continuation of #379, this PR:
- Adapts new sampler in `pulser.sampler` to extract the samples used by the `Simulation` class.
- Allows for simulation of sequences from their modulated output
- Introduces small modifications to the sampler that were missed in #388 
- Changes how noise is introduced to the samples. In particular, changes the application of amplitude noise to a more physically sound one.

Closes #315.
closed 2022-08-05 by HGSilveri