pasqal-io/Pulser #342: Sequence sampler component

lvignoli, 2022-03-04:
Samples of `Sequence` instances are currently computed in place in the `Simulation` class and in the `_seq_drawer` function as well. This gives [two](https://github.com/pasqal-io/Pulser/blob/8d0c4296348be414da1793eb3ece174422d0158b/pulser/simulation/simulation.py#L408) [pieces](https://github.com/pasqal-io/Pulser/blob/8d0c4296348be414da1793eb3ece174422d0158b/pulser/_seq_drawer.py#L32) of code dedicated to the same task, which is bad. 

To 1) fix this 2) ease the construction of the samples dictionary and 3) decouple its use from its construction, it would be very beneficial to implement an independent Sequence sampler component.
Se the [original proposal](https://github.com/pasqal-io/Pulser/issues/337#issuecomment-1059064379_) in PR #337.

The sampler will receive a `Sequence` instance as well as the strategies to sample it: noise sources, qubits masked by an SLM, modulation effects from AOM/EOM. These strategies must play well with each others: they should be easy to add, remove or update independently.

HGSilveri, 2022-03-07:
A couple more concrete ideas to consider here:
- The `SequenceSampler` (maybe we should call it this, I'll just adopt the name for now) should be initialised with just the `Sequence` instance, and then be capable of returning the samples according to the parameters given to its method(s)
- I don't know yet if it makes more sense to have a single public method to return the samples (where everything is specified in the arguments or in some configuration parameters) or to have multiple ones. If we are going to distinguish between the sampling for `_seq_drawer`, then that one would fit well in a separate method, but I don't know if much more splitting makes sense.

lvignoli, 2022-03-14:
@HGSilveri, I have opened PR #345 as a proposal implementation.

I wrote a different design that is not class-based, let me know what you think of it.

PR #345 (2022-03-28): Sequence sampler
Close #342.

A working but unfinished Sequence sampler. The tests fail on purpose: I am still thinking about the best way to test this. Also, not all noise sources have been considered.

A new module called `sampler` exposes the `sampler.sample()` function, which is given a sequence in argument and returns a nested dict of Global and Local samples per basis. The construction of this dict mimics the one in the `simulation` module.

Arguments given to `sampler.sample()` allow for modulation and noise models relevant at the sampling level.

Some comment on the internals:

- `Samp [... cut]
closed 2022-03-28 by HGSilveri