pasqal-io/Pulser #349: Sequence sampler feature extension

lvignoli, 2022-03-22:
The sequence sampler added by #345 deserves the following features

- [ ] a notebook to demonstrate its use and extend the User Guide
- [ ] multiple output format: for now it give only the nested dict form required by the simulation module
- [ ] a better way to specify the sampling-related noises
- [ ] add the possibility to specify the sampling rate

It would then be possible to use it in the `simulation` module and the `_seq_drawer()` function.

arthurfaria, 2022-07-20:
I would like to work in this issue

lvignoli, 2022-07-20:
Hi @arthurfaria! Thank you for your interest. The sampler is currently being reworked in PR #388, so this issue is not up do date.
We'll update the sampler-related issue after #388, you might be interesting in working on it after that 😃.

HGSilveri, 2022-07-20:
Very cool to know you're interested, @arthurfaria ! It is as @lvignoli says, we're currently reworking the sampler. I can always ping you once that's done.

This issues actually encompasses multiple working points, are you interested in any one in particular?

arthurfaria, 2022-07-20:
Hi @HGSilveri and @lvignoli, thanks for the reply. Well, right now I am more interested in the simple ones. I have lately grown very interested in your open-source and working on an open `good first issue` would help me very much to grasp better the ideas behind it. So among those tasks, which one do you think is the best to start with?

lvignoli, 2022-07-20:
Sure, writing an example notebook is the obvious entry point to the code.
Probably #135 is a safer route, since it does not rely on current WIP.
In any case @arthurfaria your contributions are welcomed 🙂.

arthurfaria, 2022-07-20:
alright! thanks a lot

HGSilveri, 2022-08-03:
@arthurfaria Can you please leave a comment on PR #367 so that I can assign it to you?

arthurfaria, 2022-08-03:
Of course: Hi!

HGSilveri, 2022-08-03:
> Of course: Hi!

Thanks, but can you do it on that issue #367 instead? It is more focused on what you'll be doing

HGSilveri, 2022-08-03:
Update on this issue:
- The notebook will be addressed in issue #367 
- The new sampler introduced in #388 allows for the different output formats desired
- In #390, noise and sampling rate are handled externally by the `Simulation` class (because that's the only place they're needed), so they don't need to be incorporated in the sampler.

With that being said, I think we can consider this issue concluded, so I'll close it.

closed 2022-08-03 by HGSilveri