pasqal-io/Pulser #352: SPAM errors introduce large simulation overhead

darcangelomauro, 2022-03-30:
Doing simulations with SPAM errors introduces a significant time overhead dependent on the `samples_per_run` argument, which was intended to give extra statistics (almost) for free.

The reason is that detection errors (false positives and false negatives) cycle on the sampled bitstrings and randomly flip the bits. In the following simple example execution time increases tenfold when increasing `samples_per_run` from 10 to 1000:
```python
reg = Register.square(side=2)
pulse = Pulse.ConstantDetuning(BlackmanWaveform(duration=1000, area=np.pi),0,0)
seq = Sequence(reg, MockDevice)
seq.declare_channel('ch', 'rydberg_global')
seq.add(pulse, 'ch')
sim = Simulation(seq)
noise_config = SimConfig(noise=('SPAM', 'doppler', 'amplitude'),
                             eta=0.005, epsilon=0.03, epsilon_prime=0.08,
                             temperature=30, laser_waist=148,
                             runs=10, samples_per_run=10)
sim.set_config(noise_config)
sim.show_config()
res = sim.run()
```
A first issue is that reducing the `epsilon` and `epsilon_prime` parameters to zero does not get rid of the overhead.
Secondly, the time overhead could probably be improved following a suggestion of @Louis-PaulHenry.

As @sebgrijalva pointed out, reducing the evaluation times helps dramatically, so it could be used as a workaround.

Louis-PaulHenry, 2022-03-30:
@darcangelomauro When you're mentioning a suggestion from me, do you mean computing the "reweighting" matrix?

darcangelomauro, 2022-03-30:
Yes @Louis-PaulHenry, that would eliminate the need to flip bit by bit in each string right?

Louis-PaulHenry, 2022-03-30:
yes, I think so (we can discuss it further in private).

dakk, 2023-05-31:
I think I've got it; this is the elapsed time before and after my code optimizations.
[image]

I'll prepare the commit today

PR #529 (2023-06-02): Fix SPAM errors introducing overhead
- Fixes #352 
- Handle eps_p = eps = 0 case