pasqal-io/Pulser #359: Automatize docstrings types

CdeTerra, 2022-04-11:
Currently, functions' parameters and return types are precised in the docstrings. It is redundant, because they are already given as typehints.

```python
    @_store
    def target(
        self,
        qubits: Union[QubitId, Iterable[QubitId], Parametrized],
        channel: str,
    ) -> None:
        """Changes the target qubit of a 'Local' channel.

        Args:
            qubits (Union[int, str, Iterable]): The new target for this
                channel. Must correspond to a qubit ID in device or an iterable
                of qubit IDs, when multi-qubit addressing is possible.
            channel (str): The channel's name provided when declared. Must be
                a channel with 'Local' addressing.
        """
```

It would be great to remove these repetitions, by automatically inferring the types in the docs from the typehints. We might for example try to use the library https://github.com/tox-dev/sphinx-autodoc-typehints for that purpose.

lvignoli, 2022-04-11:
That would be very cool! Does it work with how we specify the arguments or does it requires the `:param` clause?

WingCode, 2022-06-05:
@lvignoli In our function signature, we have to define the types for the arguments and the sphinx extension will pick it up from there.
@HGSilveri I would like to work on this issue. Could you assign it to me?

HGSilveri, 2022-06-05:
For sure @WingCode , sorry for the delay!

PR #376 (2022-06-17): Add sphinx typehints auto generation [unitaryhack]
closes #359
closed 2022-06-17 by HGSilveri