pasqal-io/Pulser #367: Tutorial on modulated output sequence creation and simulation

HGSilveri, 2022-05-10:
A tutorial for the docs showcasing how to use the output modulation features. 

Should be done after #315.

arthurfaria, 2022-08-03:
Hi! Thanks @HGSilveri

HGSilveri, 2022-08-03:
Awesome! 
By the way, the features needed for the simulation part are being introduced in PR #390, you can take a look already if you want. 
Relevantly here, you'll have an extra option in `Simulation()` that let's you choose whether you want to simulate with the input or the modulated output. All you'll need to do is choose this enable option, like so:
```python
sim = Simulation(seq, with_modulation=True)
```
If then e.g. you draw the simulation with `sim.draw()`, you'll see the output pulses will appear.

arthurfaria, 2022-08-03:
Thanks for the remark!

PR #421 (2022-12-15): Incorporate EOM mode operation in `Sequence`
- Adds EOM mode operation to `Sequence`
- Incorporates EOM mode modulation in the `sampler`
- Adds a tutorial on output modulation and EOM mode operation (purposefully stored without output; to check the output, see the readthedocs CI build)

Closes #367 , #418.
closed 2022-12-15 by HGSilveri