pasqal-io/Pulser #422: Inconsistent type annotation in Simulation.run()

dehond, 2022-11-28:
In the `run()` method of the `Simulation()` class there's an inconsistent type annotation for the options passed to the QuTiP solver ([see line](https://github.com/pasqal-io/Pulser/blob/develop/pulser-simulation/pulser_simulation/simulation.py#L819)). It says that the type should be a `qutip.solver.Options` object, but the code looks like it's treated as a dictionary, with the key-value pairs being splatted into the `Options` object when it's generated [a few lines later](https://github.com/pasqal-io/Pulser/blob/develop/pulser-simulation/pulser_simulation/simulation.py#L847).

No matter how I try, I cannot seem to get it to accept a QuTiP `Options` object, but things are fine with a regular `dict`.

HGSilveri, 2022-11-28:
Hi @dehond , you're right... It expects the arguments to `Options`, not an `Options` instance. 
Thanks for bringing this up, it should be very simple to fix. Would you like to do it or should we go ahead and do it ourselves?

HGSilveri, 2022-11-28:
Awesome, @dehond, the issue is yours!

PR #423 (2022-11-29): Fix type annotation in run() method of Simulation() class.
Fixes #422