pasqal-io/Pulser #427: Update resolution method for Quantum Master Equation [`pulser-simulation`]

sebgrijalva, 2022-12-05:
When calling the `run()` method in `Simulation` (line 880 in `Simulation.py`), a fix was introduced for using the `mesolve()` method from QuTiP, in which a `Liouvillian` object was created before calling the solver.

In the meantime, the creation of the Hamiltonian and initial state was updated in QuTiP, so the "fix" should be updated with the recommended creation.

sebgrijalva, 2022-12-05:
@Varda-star  This is one of the improvements on the module for you :)

Varda-star, 2022-12-06:
I jump on it!

Varda-star, 2022-12-09:
In the meantime, the creation of the Hamiltonian and initial state was updated in QuTiP, so the "fix" should be updated with the recommended creation. // What are these updates precisely?

sebgrijalva, 2022-12-12:
Don't create the `qutip.liouvillian()` object in line 880, instead call the `mesolve()` method directly with the Hamiltonian and collapse operators.

Varda-star, 2022-12-12:
Copied that

Varda-star, 2022-12-12:
I open a PR about it.

PR #434 (2022-12-27): Update method for Quantum Master Equation
Fixes #427
closed 2022-12-27 by HGSilveri