pasqal-io/Pulser #428: Generalize effective noise channels

sebgrijalva, 2022-12-06:
So far we have added a dephasing channel for noisy `Simulation`. We can now generalize it to a depolarizing channel and more generally to any custom type of "collapse" operators for the Lindblad equation solver. Please look at QuTiP documentation [here](https://qutip.org/docs/latest/guide/dynamics/dynamics-master.html).

sebgrijalva, 2022-12-06:
@Varda-star This is for after you are done with #427 , since you will be already familiar with the structure of the Simulation module.

Varda-star, 2022-12-12:
Do I build this operator pulsewise or using Qutip?

Varda-star, 2022-12-12:
Does the restrictions for the dephasing Noise apply for the Generalize effective noise channels? 
For instance (the exception " Cannot include dephasing noise in digital- or all-basis.")

Varda-star, 2022-12-12:
Why the dephasing doesn't happen in the digital mode?

sebgrijalva, 2022-12-12:
> Do I build these operators pulsewise or using Qutip?

Preferably using `QuTiP`'s operators for the Lindblad collapse operators.

> Why the dephasing doesn't happen in the digital mode?

It can be added, but then you have to be extra careful about the form of the collapse operators on the digital and 'all' basis (which are qutrits). Probably doing the allowed evolutions by pieces and building the right projectors. 

I believe we decided to only implement this for the `ground-rydberg` basis because it's where an effective model is most meaningful. 

For the moment let's add the depolarizing channel only for `ground-rydberg`(please be extra careful about the construction of the many-body collapse operators, convince yourself you agree with lines 219-225 for creating them)

Then the general operators are supposed to be given as input to the Lindblad equation. But we should think carefully if there is some checking to do first. For this second part, I suggest pasting here a minimally working example script that we can use to discuss before going for the full code.

Varda-star, 2022-12-14:
Copied that. The first PR I did is being reviewed, meanwhile I will start adding the depolarizing channel, write a UT.

Varda-star, 2022-12-14:
[image]
Can I add `sigma_rg` to the op_ids in the `ground_rydberg` basis?

PR #443 (2023-01-19): Generalize effective noise channels
Fixes #428
closed 2023-01-19 by HGSilveri