pasqal-io/Pulser #454: pulser-core 0.8.0 pins jsonschema to 4.4.0

picbeats, 2023-01-13:
pulser-core 0.8.0 pins jsonschema to 4.4.0 causing conflicts with other packages due to missing `format-nongpl` extension. 
Removing of the constrain does not make any test fail. 
Suggested to remove jsonschema constrain.

HGSilveri, 2023-01-13:
Hi @picbeats , thanks for raising this up!
This seems sensible, as far as I can remember we don't have a strong reason to pin the `jsonschema` version. 
Is this urgent or have you managed to circumvent this issue for now?

picbeats, 2023-01-16:
It's not urgent, but required a workaround in a build pipeline, so would be nice to fix it with a next release.

HGSilveri, 2023-01-17:
Alright, no problem!

closed 2023-01-18 by HGSilveri

HGSilveri, 2023-01-18:
@picbeats The change has been released in v0.9.0, feel free to update your pipeline.

picbeats, 2023-01-18:
Thanks, that's worked. Still have issues with requests and pasqal-sdk vs. jupyterlab-server

PR #456 (2023-01-18): Delete pinning jsonschema to version 4.4.0
I have deleted the requirement `jsonschema==4.4.0`.

Closes #454