pasqal-io/Pulser #478: Exception handling - raise more consistent errors in deserialize_device

Augustinio, 2023-03-03:
In pulser/json/deserializer:

```
def deserialize_device(obj_str: str) -> BaseDevice:
    """Deserialize a device from an abstract JSON object.

    Args:
        obj_str: the JSON string representing the device encoded
            in the abstract JSON format.

    Returns:
        BaseDevice: The Pulser device.
    """
    obj = json.loads(obj_str)
    # Validate the format of the data against the JSON schema.
    jsonschema.validate(instance=obj, schema=schemas["device"])
    return _deserialize_device_object(obj)
    ```
    
    There are multiple ways this function can fail, and raising multiple different classes of Error.
    We use this function to validate a serialised abstract representation of a device. Because of the unpredictability of the exceptions raised in this function, we have to cast a wide net with a `try ... except Exception... ` which is less than ideal.
    
    - First point is obviously the json load that could fail and that is not caught. (could raise `TypeError` or `JSONDecodeError`)
    
    - Second point the the `jsonschema.validate(instance=obj, schema=schemas["device"]` function that can fail in multiple ways: at least:
    ```
     `jsonschema.exceptions.ValidationError` if the instance
            is invalid

        `jsonschema.exceptions.SchemaError` if the schema itself
            is invalid
   ```
   according to the documentation. Though I've not dug deeper to see if other uncaught exceptions could arise.
   
   - Lastly the _deserialize_device_object, which I suspect could also throw other errors, especially when registering the layouts.
   
   Ultimately what would be nice is a unique Exception class that we could catch for when using this deserialize_device function.

MatthieuMoreau0, 2023-03-03:
I see there is an [AbstractRepresentationError](https://github.com/pasqal-io/Pulser/blob/4592d39765e6c864f35d264bbf0ccb182562bf96/pulser-core/pulser/json/exceptions.py#L23) but it looks like it is only used when serializing to the abstract representation.

Would it be possible to have an equivalent error raised consistently on deserialization ?

HGSilveri, 2023-03-03:
@Augustinio If I understand correctly, what you are looking for is that I group all possible errors that may arise from an invalid argument to `deserialize_device()` under the same exception. If so, then errors like `jsonschema.exceptions.SchemaError` should not be contemplated, right?

> I see there is an [AbstractRepresentationError](https://github.com/pasqal-io/Pulser/blob/4592d39765e6c864f35d264bbf0ccb182562bf96/pulser-core/pulser/json/exceptions.py#L23) but it looks like it is [...]

`AbstractRepresentationError` is already being used by all errors raised directly by the deserializer. What you are suggesting is that we catch all others too and turn them into an `AbstractRepresentationError`?

This seems feasible, when would you need this?

Augustinio, 2023-03-06:
I guess it depends on how you want to structure your error classes internally. You could have a parent `DeserializationError` from which some of your errors inherit. Or you could catch any of the 5 or 6 exception classes you currently raise and then raise a `DeserializationError`. But yeah the core requirement is to be able on our side to do a `try deserialize except SomeError`.

As far as how soon we need it, it is not crucial at the moment as we just catch any `Exception`. So if you want to take some time to think of the best way to structure your exceptions, there's no rush.

HGSilveri, 2023-03-06:
> I guess it depends on how you want to structure your error classes internally. You could have a parent `DeserializationError` from which some of your errors inherit. Or you could catch any of the 5 [...]

Ok. So, in principle, it would be okay if this feature was treated normally (ie added to the `develop` branch and eventually release on the next Pulser release)? In other words, would it be okay for you to then update `pulser-core` to a new minor release?

MatthieuMoreau0, 2023-03-09:
Yes this would be completely fine!

PR #482 (2023-03-29): Implement custom exception for device deserialization
- Creates `DeserializeDeviceError` and makes it the only error raised by `deserialize_device()`
- Adds an additional check to `RegisterLayout()` to forbid duplicates in trap coordinates

Fixes #478 .
closed 2023-03-29 by HGSilveri