pasqal-io/Pulser #479: Allow to customize the colors when drawing a register

CdeTerra, 2023-03-08:
Currently, when we plot a register, they are all shown in green. Just like it's possible to do some plotting customization, as specifying masked qubits, I would also like to be able to specify individual colors for each qubit.

The signature of `Register.draw` and `RegDrawer._draw_2D` would be added a new parameter:

```python
qubit_colors: dict[QubitId, str]
```

This parameter would basically apply where "darkgreen" currently does, hence the point of the qubit as well as the transparent disk and the circle when the "draw_half_radius" is set to `True`.

HGSilveri, 2023-03-09:
Does the user need to specify a color for every qubit, or can does it default to "darkgreen"?

CdeTerra, 2023-03-09:
I think it depends on whether the current default darkgreen color is meant to stay. If so, we can allow the user not to specify the colors of all atoms. Otherwise, it's better to enforce the user to specify all of them, so that if some day the default color changes, it won't overlap the marking color they will have chosen.

HGSilveri, 2023-03-09:
I'm not sure I follow that logic. What I'm suggesting is that whenever the user chooses not to specify the color of a qubit, we use a default. Even if that default changes, it will never overlap the color of qubits they have chosen.

In practice, it would be something like this:
```python
def default_color():
    return "darkgreen"

all_qubit_colors = defaultdict(default_color, qubit_colors)
```

CdeTerra, 2023-03-09:
What I mean is that for example, if the user decides to mark a few qubits to orange, and that later on Pulser switches the default color to orange, the mark of the qubits won't be visible anymore. But if the user precises all colors (orange and dark green for example), even if the default color of Pulser changes, marks will still be distinguishable.

CdeTerra, 2023-03-09:
Okay, it's all good for me. Is there any other point you would like to discuss about before I start on this?

PR #480 (2023-03-10): Enable to choose custom colors of qubits when drawing Register
Closes #479 

Example of use:

```python
from pulser.register import Register
r = Register.from_coordinates([[0,0], [10,0], [15,3], [2,4], [5, -3]])
r.draw(
    blockade_radius=8,
    draw_half_radius=True,
    qubit_colors={3: "darkred", 2: "darkred", 4: "#45b9fe"}
)
```

[image]

@HGSilveri : should I also apply this to `Register3D` to keep consistency?
closed 2023-03-10 by HGSilveri