pasqal-io/Pulser #484: [FIX] serialisation of devices with pre_calibrated_layouts

Augustinio, 2023-03-20:
I'm realising that the serialization of a device with calibrated layouts does not work.

`_to_abstract_representation() `returns a dict with all the values properly serialized except for  `RegisterLayout` in the list of `pre_calibrated_layouts`. For instance:

```
{'version': '1', 'channels': [], 'name': 'Chadoq2', 'dimensions': 2, 'rydberg_level': 69, 'min_atom_distance': 0.01, 'max_atom_num': 10, 'max_radial_distance': 5, 'interaction_coeff_xy': None, 'supports_slm_mask': False, 'max_layout_filling': 0.5, 'reusable_channels': False, 'pre_calibrated_layouts': (RegisterLayout_e24e4ba29bf5f7fcf190dd1a1022d1757d2d2792651d61a5707bb5b7e5a285cf,), 'is_virtual': False}
```

`RegisterLayout` is of course not json serializable. So we can't save the specs as a json. But we also couldn't deserialize them with the method from the lib because it contains `RegisterLayout` instances. However the `deserialize_device` method does account for potential `pre_calibrated_layouts`:

```
def _deserialize_device_object(obj: dict[str, Any]) -> Device | VirtualDevice:
    device_cls: Type[Device] | Type[VirtualDevice] = (
        VirtualDevice if obj["is_virtual"] else Device
    )
    ch_ids = []
    ch_objs = []
    for ch in obj["channels"]:
        ch_ids.append(ch["id"])
        ch_objs.append(_deserialize_channel(ch))
    params: dict[str, Any] = dict(
        channel_ids=tuple(ch_ids), channel_objects=tuple(ch_objs)
    )
    ex_params = ("_channels", "channel_objects", "channel_ids")
    for param in dataclasses.fields(device_cls):
        if not param.init or param.name in ex_params:
            continue
        if param.name == "pre_calibrated_layouts":
            key = "pre_calibrated_layouts"
            params[key] = tuple(
                _deserialize_layout(layout) for layout in obj[key]
            )
        else:
            params[param.name] = obj[param.name]
    return device_cls(**params)
```

I've also notice that there is a `_to_abstract_representation()` method on the `RegisterLayout` object. So this should be an easy fix. On the `_to_abstract_representation()` of the `Device` model:

```
def _to_abstract_repr(self) -> dict[str, Any]:
        d = super()._to_abstract_repr()
        d["is_virtual"] = False
        if self.pre_calibrated_layouts:
            d["pre_calibrated_layouts"] = tuple(
                layout._to_abstract_repr() for layout in self.pre_calibrated_layouts
            )
        return d
```

This is a fairly pressing issue as it is blocking the implementation of device specs serialisation for us.

awennersteen, 2023-03-22:
Isn't it confusing that for the sequence we can (and do) serialise with `seq.to_abstract_repr` but we cant for the device?

Is there any reason why such behavior was dropped for the device? or is it also discouraged do use `seq._to_abstract_repr`?

HGSilveri, 2023-03-22:
> Isn't it confusing that for the sequence we can (and do) serialise with `seq.to_abstract_repr` but we cant for the device?
> 
> Is there any reason why such behavior was dropped for the device? or is it also discouraged do use `seq.to_abstract_repr`?
> 
> (I do note that the device method has the "private method underscore", and not the sequence. However, this is to me still not sufficient signaling)

There is nothing preveting us from adding a `Device.to_abstract_repr()`, the only reason I didn't was because it wasn't something I expected regular users to need. I figured that internally, it would be sufficient to dump with the `AbstractReprEncoder`.
That being said, I'd be happy to add this method, even if it's just for being consistent and avoiding misunderstandings (though it seems it will be useful as well).

HGSilveri, 2023-03-22:
UPDATE: I forgot this had been defined, but actually all devices have a `to_abstract_repr()` method that should be the default way of serializing them. It is defined on the parent class (`BaseDevice`). 

So, for a given device instance, running `device.to_abstract_repr()` should give you a correct JSON-formatted string.

closed 2023-03-22 by HGSilveri