pasqal-io/Pulser #496: Adding effective channel noises to digital basis

a-corni, 2023-03-30:
Currently, effective channel noises are implemented only for the Rydberg basis. A `NotImplementedError` is raised when a simulation is ran with a `Simconfig` having `detuning` or `dephasing` noise.
Implementing this feature would have an impact on simresults: the reduction of a density matrix to a basis is not implemented and cannot be tested because of the fact that no simulation can be performed with dimensions 3. This should be performed using a [partial_trace](https://physics.stackexchange.com/questions/179671/how-to-take-partial-trace)

Varda-star, 2023-04-13:
The idea is it to build effective channels on 3 dimensions then reduce it in the basis of interest using the partial trace?
What would be the interest to work with the digital basis?

HGSilveri, 2023-04-13:
Yes, basically because it can be interesting to see how noise affects the results in digital mode.

Aaron-Robertson, 2023-05-30:
This one looks enjoyable! I'm giving it a shot, and expect a draft PR today. I'll likely ask for guidance at that point

HGSilveri, 2023-05-30:
@Aaron-Robertson awesome, looking forward to it!

Aaron-Robertson, 2023-05-31:
I'm getting there @HGSilveri! It's been a more intense learning experience than anticipated 🤣 but I should actually have a draft today!

HGSilveri, 2023-06-01:
Let me add a bit more clarification here. The objective is that we want to generally run simulations with effective noise added to the system. Right now, this is only supported for the "ground-rydberg" basis, though we could envision it being extended to the "digital" basis relatively easily. 
However, in the "all" basis, the state of each atom in our system is actually described by **three-level system** (aka a qutrit), which means that the collapse operators that we have for qubits need to be extended to qutrits. This is the crux of the problem, as we don't know exactly what this entails. Once we know how to do this extension, it should "just" be a matter of constructing the collapsed operators depending on the basis.
So, this requires some preparation before the implementation. In particular, it requires figuring out what the Kraus operators for simulatenous dephasing and/or depolarization on a qutrit combining the ground-rydberg and digital basis.

HGSilveri, 2023-09-27:
Most of the work on this has already been done in PR #530. That PR has since been closed due to prolonged activity, but whoever picks this up in the future can base themselves on the work already done there.

closed 2024-08-07 by a-corni