pasqal-io/Pulser #497: Add an error when a `Channel` is called with an `eom_config` but without a `modulation bandwidth`

a-corni, 2023-04-05:
No error is raised when a `Channel` is called with an `eom_config` but without a `modulation bandwidth`.
This is currently only the case for a `Rydberg` channel. It should be generalized to all channels.

atomgardner, 2023-05-29:
Oh—it looks like I was beaten to the punch by @dakk. @HGSilveri, I'll be sure to follow the CONTRIUTING.md advice in future :smile:

HGSilveri, 2023-05-30:
Hi @dakk and @atomgardner ! First off, thank you both for your contributions 😄 
Since you both have identical changes and failing CI tests, we have decided to give the prize to the one who gets all CI tests passing first!

@atomgardner : For the sake of fairness, note that @a-corni already gave @dakk some hints on how to fix his failing tests.

a-corni, 2023-05-30:
Thank you both for your contributions ! @dakk could you please comment here such that I can assign you the issue ?

dakk, 2023-05-30:
> Thank you both for your contributions ! @dakk could you please comment here such that I can assign you the issue ?

Sure!

a-corni, 2023-05-30:
Thank you @dakk and @atomgardner for your participation ! I am looking forward seeing your contributions to other issues !

PR #526 (2023-05-30): Add an error when a Channel is called with an eom_config but without …
…a modulation bandwidth

Fixes #497
closed 2023-05-30 by a-corni