pasqal-io/Pulser #501: Drawing of `Register` goes out of window boundaries

LouisJustinTALLOT, 2023-04-10:
When plotting a very simple register, the axes of the plot are going out of the boundaries of the figure, making it unreadable without manual resizing : 

[image]

Code to reproduce the bug : 

```python
from pulser import Register

register = Register.rectangle(rows=1, columns=2)

register.draw()

```

This issue appears on the following platforms and modules versions : 

| Platform           | Python | Pulser | Matplotlib |
| ------------------ | ------ | ------ | ---------- |
| Windows 10         | 3.10.9 | 0.10.0 | 3.7.1      |
| Ubuntu 20.04 WSL 2 | 3.10.9 | 0.11.0 | 3.7.1      |

HGSilveri, 2023-04-11:
Hey @LouisJustinTALLOT , thanks for the suggestion! It's definitely something we can do.

AmanSal1, 2023-04-12:
@LouisJustinTALLOT Also if you notice one thing if the row number is kept equal to 1 then only this problem occurs but if we keep it greater than 1 it then works fine irrespective of the columns we put.

dakk, 2023-05-30:
Sure @a-corni . The solution I am proposing should work for every chart