pasqal-io/Pulser #511: Minor release to incorporate recent pasqal-cloud changes?

ccalaza, 2023-04-28:
Dear devs,

I'm currently trying to install Pulser on our HPC systems at Forschungszentrum Juelich. 

There was some incompatibilities between `pasqal-cloud` requirements and the software we have in production at our site (for details see https://github.com/pasqal-io/pasqal-cloud/issues/75).

These issues have been very quickly resolved by your colleagues, but when I attempt to install Pulser again it seems like the current latest Pulser release in PyPi still points to an old version of `pasqal-cloud`, prior to the fix I needed.

Would it be possible to post a minor update of Pulser to Pypi so that I can carry on with the installation on our systems?

Many thanks in advance!

Best regards,
Carlos Gonzalez
JUNIQ - JSC - Forschungszentrum Juelich

HGSilveri, 2023-04-28:
Hi Carlos!

I think we can get a release out by Wednesday next week (potentially Tuesday), would this work for you?

ccalaza, 2023-04-28:
Sounds great, thanks a lot!

HGSilveri, 2023-05-02:
Hey @ccalaza !

v0.12.0 has just been released, so hopefully you'll be able to proceed with your installation.

closed 2023-05-02 by HGSilveri

ccalaza, 2023-05-03:
Thanks a lot, it worked!