pasqal-io/Pulser #551: Pulser symbols exports not friendly to pyright

lvignoli, 2023-07-07:
Many symbols that are effectively _exported_ by pulser are _not discovered_ by pyright.
This is the case for `pulser.Sequence` or `pulser.Register` for instance.

[image]

I suspected that other static analyser may experience the same problem. This is inconvenient when developing, as one looses autocompletions and suggestions.

I am unsure on the reason of this, but 
- pyright reports some `reportPrivateImportUsage` on `pulser-core/pulser/__init__.py`. I guess this is related to ruff F401s
- There are no dunder `__all__` in any of pulser `__init__.py`, apparently unrelated but who knows.

lvignoli, 2024-01-08:
Any update in this direction? This heavily cripples the developper experience when using Pulser as a dependency. VSCode is impacted as well, so it's very unfriendly to beginners.

HGSilveri, 2024-01-08:
Sorry for the dumb question but I still need to ask: is your VSCode interpreter pointing to your venv folder where pulser is installed? 
[image]

HGSilveri, 2024-01-08:
Ok, if I manually install `pyright` and execute it, your error pops up. This should be enough for me to work on this.
I still wonder, is pyright enabled by default on VSCode? I might have personalized settings for this, I'm not sure, but I can't find any mention of pyright on my installation and everything seems to work fine. I'll still try to fix this but it makes me wonder if this is not a `pyright` issue... TBD, I guess

lvignoli, 2024-01-09:
> Sorry for the dumb question but I still need to ask: is your VSCode interpreter pointing to your venv folder where pulser is installed? [...]

Yes I am working in the correct venv 😉

lvignoli, 2024-01-09:
In VSCode the default out-of-the-box python experience is provided by Pylance, which builds on top of pyright.
So any default VSCode setting actually make use of pyright.

It is possible that pyright implements an unorthodox way for deciding what symbols is actually exported. Maybe by implementing a PEP that no one actually respect?

lvignoli, 2024-01-09:
I was not able to install your branch as a deps locally, but reproducing your changes does correctly expose the symbol.
This is apparently the intended verbose way to tell type checkers what is exported, this or using `__all__`,see https://github.com/microsoft/pyright/blob/main/docs/typed-libraries.md#library-interface.

Another fix is simply to remove `py.typed`, marking the library as not typed. I have tried and it works out of the box. I don't understand the rationale of making it harder to export when you declare your library as typed, but okay…

lvignoli, 2024-01-09:
For reference:
- no `py.typed`: [Pennylane](https://github.com/PennyLaneAI/pennylane/tree/master/pennylane)
- use `__all__`: [PyTorch](https://github.com/pytorch/pytorch/blob/db79ceb110f6646523019a59bbd7b838f43d4a86/torch/__init__.py#L41), [numpy](https://github.com/numpy/numpy/blob/f0a8e0807f42c67cd85d8ddd3bbbb82b1dbff695/numpy/__init__.py#L296)

HGSilveri, 2024-01-09:
As much as I'd like to just remove `py.typed` and avoid the hassle of using `__all__`, I can't do it in good conscience...

HGSilveri, 2024-01-10:
Ok, I have question on this: how much do we want to expose? It's clear that the symbols directly imported (eg `Sequence`, `Pulse`, ...) should be exposed, but what about submodules?
To me, it makes sense to expose submodules like `devices` and `waveforms`, but most are used only internally. Some, like `pulser.json` are clearly internal only (and should have probably been private from the start...) but others (eg. `channels`) I guess it might depend: for a user, I would say they should be hidden, but for a developer having the autocomplete on `pulser.channels.[...]` might come in handy. 
Do you have any insight on this?

lvignoli, 2024-01-11:
Indeed, this asks the question of what it is exactly the public API of pulser. For sure developers want more such as devices and waveforms and channels and samplers …
To me, a simple answer is everything already documented [here](https://pulser.readthedocs.io/en/stable/apidoc/pulser.html) must be exported.

lvignoli, 2024-01-11:
I would also make a case for something that I advocate for a long time: _exporting everything under the single `pulser` top-level namespace._

Everyday pulser usage requires lots of boilerplate imports of naked names.
A single import line mitigates this, and deletes the need to search for the correct module path.
Pulser is not so big yet, so we could totally afford to have the 30-ish everyday useful symbols in a single namespace.

To support this, it seems most of the symbols have been named in a way that allows and even compels to do so:
`MockDevice` is redundant—being in the `device` module it could have been named `Mock` and used as `device.Mock`. But since the name carries context, we could export it at top-level to be called as `pulser.MockDevice`, and consider the `device` module just as an internal way to organize code.

This works for
- devices: `MockDevice`, `AnalogDevice`, etc.
- waveforms: `ConstantWaveform`, `CompositeWaveform`, etc.
- results: `SampledResults`, `QutipResults`, etc.
- backends: `QPUBackend`, `QutipBackend`, etc.

HGSilveri, 2024-01-11:
> I would also make a case for something that I advocate for a long time: _exporting everything under the single `pulser` top-level namespace._

I'm already playing around with the best way to do this on this PR. To me, the answer here is relatively clear: we want the top-level namespace to give access to everything a regular user needs and nothing more. This would exclude things like abstract base classes or even classes the user interacts with but never needs to initialize (eg. `SampledResults` et al). 
I think this decision then somehow conflicts with your other point:

> To me, a simple answer is everything already documented [here](https://pulser.readthedocs.io/en/stable/apidoc/pulser.html) must be exported.

Many classes we include in the API documentation are there so the users can have easy access to their methods' signature and docstring, but they should not be instantiated in isolation. I feel like exposing them sends the wrong message.

lvignoli, 2024-01-11:
Indeed, I have suggested that a bit too quickly.

> To me, the answer here is relatively clear: we want the top-level namespace to give access to everything a regular user needs and nothing more. This would exclude things like abstract base classes or even classes the user interacts with but never needs to initialize (eg. SampledResults et al).

I fully stand by your side with this. I trust your opinion on what should make it to the list and I'll give you feedback if needed.

Thanks a lot for considering fixing the original issue with pyright but also rationalizing the user experience! 😃

closed 2024-01-23 by HGSilveri