pasqal-io/Pulser #577: Add function to validate that a given sequence's register matches the pre calibrated layout of a device

MatthieuMoreau0, 2023-09-06:
Given a `Sequence` and a `Device` object, I would like to be able to assert that the register of the sequence derives from the pre calibrated layouts of the device (if any). If no pre-calibrated layouts on the device then the function would return always True.

HGSilveri, 2023-09-06:
> If no pre-calibrated layouts on the device then the function would return always True.

I'm not sure about this part... Imo, if you're trying to check that a register came from a layout in a Device's pre-calibrated layouts but that Device has no pre-calibrated layouts, I would expect either `False` or an error, not `True`.

MatthieuMoreau0, 2023-09-06:
> I'm not sure about this part... Imo, if you're trying to check that a register came from a layout in a Device's pre-calibrated layouts but that Device has no pre-calibrated layouts, I would expect either False or an error, not True.

Yeah I see how this is confusing, let's go with False in that case!

> Ok, so you'd prefer that the input to the method is the register, not the layout?

If possible yes, this would make my life marginally easier as I wouldn't have to explicitly check if the register has a layout before running that function

HGSilveri, 2023-09-20:
My idea for this was to add two new methods to `Device`:

- [ ] `Device.is_calibrated_layout(layout: RegisterLayout)`: Checks whether a layout is within the calibrated layouts.
- [ ] `Device.register_is_from_calibrated_layout(register: BaseRegister)`: Checks whether the register was constructed from a pre-calibrated layout (ie checks if it has a layout AND `is_calibrated_layout(layout)`)

This way, we can have the possibility to check whether a layout is calibrated while also having the convenience of checking directly from the register.

PR #586 (2023-09-22): Adding register_is_from_calibrated_layout and is_calibrated_layout to Device
As suggested by @HGSilveri:
Add two new methods to `Device`:
- [ ] `Device.is_calibrated_layout(layout: RegisterLayout)`: Checks whether a layout is within the calibrated layouts.
- [ ] `Device.register_is_from_calibrated_layout(register: BaseRegister|MappableRegister)`: Checks whether the register was constructed from a pre-calibrated layout (ie checks if it has a layout AND `is_calibrated_layout(layout)`)
As discussed, returns a boolean and not an error.
register_is_from_calibrated_layout works with BaseRegister (he [... cut]
closed 2023-09-22 by a-corni