pasqal-io/Pulser #607: Remove Chadoq2 device

madagra, 2023-11-08:
The `Chadoq2` device is currently deprecated and it should be removed.

HGSilveri, 2023-11-29:
Rather than removing `Chadoq2` altogether - which would be quite disruptive to existing code - I propose the following steps:

- [ ] Create a `DigitalAnalogDevice` with identical specs to `Chadoq2`
- [ ] Replace usage of `Chadoq2`with `DigitalAnalogDevice` throughout the codebase (including tests and tutorials)
- [ ] Raise a `DeprecationWarning` in `Sequence's` created with `Chadoq2`, recommending the usage of `DigtalAnalogDevice` instead (we can do the same with `IroiseMVP`)

@madagra @a-corni Do you agree with these steps? Would a different name perhaps be better? I also thought of `HybridDevice` which is shorter but vaguer

madagra, 2023-11-30:
Let's go ahead with `DigitalAnalogDevice`.

PR #620 (2023-12-01): Deprecate legacy devices
Closes #607 .
closed 2023-12-01 by HGSilveri