pasqal-io/Pulser #624: Request `pulser>=0.16.0` to support  `jsonschema==4.17.3`

ccalaza, 2023-12-12:
Dear Pulser devs,

at Forschungszentrum Juelich we are currently trying to install the current `pulser==0.16.0` version so that we can include it in our 2024 HPC SW stage. 

It seems that all your `pulser > 0.14.1` pin their `jsonschema > 4.18`, but unfortunately our SW stage provides `jsonschema==4.17.3`. I've noticed that `jsonschema` isn't used too widely by Pulser, so is there any chance you could consider supporting our `4.17.3` version? 

Many thanks in advance and sorry for the inconvenience! 
X
Carlos

a-corni, 2023-12-13:
Hello Carlos,
Thanks for your interest in Pulser ! It seems feasible, I have just opened a PR to support v4.17.3 of jsonschema.
Best regards,
Antoine

HGSilveri, 2023-12-14:
Hey @ccalaza , `pulser == 0.16.1` has been released with support of `jsonschema == 4.17.3`. Feel free to let us know if any issues persist.

closed 2023-12-14 by HGSilveri

ccalaza, 2023-12-14:
Many thanks for the quick reaction, Henrique and Antoine! 

This was the only incompatibility we spotted so I think we are good :)