pasqal-io/Pulser #658: Pulser Docs Revamp

HGSilveri, 2024-03-20:
- [ ] Improve the "Quickstart" tutorial
- [ ] Clarify how to choose a device and create a register for different applications
- [ ] Make a tutorial with the whole pipeline specifically for executing a sequence on a remote QPU
- [ ] Remove parts of (or entire) tutorials that are not relevant anymore
- [ ] Review API reference docs, fix typos and formatting bugs and make sure all public interfaces are properly documented

a-corni, 2024-06-05:
I would also add:
- [ ] Refresh application notebooks: Antiferro, 2-qubit gates, ...
- [ ] Add new application notebooks

a-corni, 2025-07-01:
- Refresh application notebooks + add new applications (MWIS) was done by #821 
- New Backend feature is now documented, will be finalized by the person tackling #875 
- Add new noise types is being tackled by #893

closed 2025-07-04 by a-corni