pasqal-io/Pulser #679: Pulser update submodules (pydantic 2.6.0 issue: 114), Forschungszentrum Jülich

AlvaroHuanay, 2024-04-23:
Greetings,

We come from the previous issue: [Pasqal-Cloud dependency with Pydantic (Forschungszentrum Jülich) #114](https://github.com/pasqal-io/pasqal-cloud/issues/114). In such incident we requested a new release of pasqal-cloud that could handle pydantic versions >= 2.6.0 which you fixed (Thank you very much for the quick action).

This week  compiling your libraries for our HPC systems we encountered an issue at the requirements of [pulser-pasqal library](https://github.com/pasqal-io/Pulser/blob/develop/pulser-pasqal/requirements.txt). All its versions (from 0.16.0 to 0.17.4) require pasqal-cloud ~=0.4.0 version. We would appreciate a new release of Pulser and pulser-pasqal that works with your newly generated pasqal-cloud => 0.7.1

Could you please upload a version such that works with your pasqal-cloud >=0.7.1?

Thank you very much.

Kind regards.

HGSilveri, 2024-04-24:
Hey @AlvaroHuanay ! Sure, we'll make a release soon. I'll let you know when it's done.

AlvaroHuanay, 2024-04-25:
Thank you!

Have a nice day.

PR #681 (2024-04-25): Update pasqal-cloud to ~=0.8.1
Closes #679 .
closed 2024-04-25 by HGSilveri