pasqal-io/Pulser #691: Implement depolarizing and dephasing operations with projectors

a-corni, 2024-06-06:
Currently, depolarizing and dephasing operations are defined in pulser_simulation/hamiltonian.py/Hamiltonian._build_collapse_operators using `qutip.sigmax/y/z()`. It would be better to define them with projectors `sigma_{a}{b}` with a, b in [r, g, h].

HGSilveri, 2024-07-22:
I started working on this but now I'm wondering if it actually makes sense to do... Can you remind me of the motivation behind this suggestion?
Doing this will complicate the code (we have to define `sigma_x/y/z` for each basis) and I'm not sure we gain much from it, since this is internal code where the Pulser convention is well established and consistent.
So, my question is: do we really need this?

a-corni, 2024-07-22:
That was to enable the definition of "depolarizing"/"dephasing" channels with the "leakage" noise option.
At first glance, I don't see why it would make it harder... Why don't we define `sigma_gg - sigma_rr` for dephasing_rate, `sigma_gg - sigma_hh` for hyperfine_dephasing_rate ? (Is this wrong with leakage noise ?)
I don't know how you would do it for depolarizing however...

HGSilveri, 2024-07-22:
It's not harder, just more verbose for no reason I could remember... If you need it, that's good enough for me

PR #715 (2024-07-29): Defining dephasing and depolarizing operators with projectors
Fixes #691 .