pasqal-io/Pulser #698: Several errors with conda/mambaforge/miniforge

Yoric, 2024-06-24:
Steps to reproduce:

1. Install a fresh copy of [miniforge](https://github.com/conda-forge/miniforge/releases) (tried on Ubuntu Linux)
2.  Activate your miniforge with `$ . ~/mambaforge/bin/activate`.
3. Install pulser with `$ pip install pulser`.

Outputs

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
tensorboard 2.15.1 requires markdown>=2.6.8, which is not installed.
```

After this, 

```python
from pulser import Pulse, Sequence, Register
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'DigitalAnalogDevice' from 'pulser.devices' (/home/XXX/.local/lib/python3.10/site-packages/pulser/devices/__init__.py)
```

HGSilveri, 2024-06-24:
Thanks for reporting @Yoric ! I just tried reproducing the bug with the steps you outlined:
1. I installed miniforge [using this command](https://github.com/conda-forge/miniforge?tab=readme-ov-file#unix-like-platforms-mac-os--linux) (also on Ubuntu 22.04)
2. `$ . ~/mambaforge/bin/activate` does not work because `~/mambaforge` does not exist. Instead, it is called `~/miniforge3` and `$ source /miniforge3/bin/activate` works. 
3. `$ pip install pulser` works without errors.
4. The pulser imports also go through without a hitch. 

So, I was unable to reproduce your errors... Our discreprancies in point 2. suggest that you installed `mambaforge` instead of `miniforge3` which,[ according to their README](https://github.com/conda-forge/miniforge?tab=readme-ov-file#mambaforge), **is discouraged as of September 2023**. Could you please install `miniforge3` and let me know if the error persisted?

HGSilveri, 2024-06-25:
@Yoric Any news on this?

Yoric, 2024-06-26:
Thanks for the suggestion.

I still encounter

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
tensorboard 2.15.1 requires markdown>=2.6.8, which is not installed.
```

with miniforge3.

However,
```py
from pulser import Pulse, Sequence, Register
```

now works.

HGSilveri, 2024-06-28:
@Yoric The dependency conflict seems to be unrelated to Pulser, so I'm closing the issue

closed 2024-06-28 by HGSilveri