pasqal-io/Pulser #699: Cannot import Device or BaseDevice

Yoric, 2024-06-26:
I'd like to write some code that takes as argument a device. Currently, both `Device` and `BaseDevice` are defined in `_device_datacls`, which marks them as private. Is there any better way to do this? If not, could either `Device` or `BaseDevice` be officially exported somewhere?

HGSilveri, 2024-06-26:
`Device` and `VirtualDevice` are officially exported under `pulser.devices`, so you can import them from there. `BaseDevice` is an ABC and is purposefully not exported, as it is intended to be an internal class only. Why would you like to export it?

Yoric, 2024-06-26:
Mmmh... Pyright informs me that:

```
"Device" is not exported from module "pulser.devices"
  Import from "pulser.devices._device_datacls" instead
```

Yoric, 2024-06-26:
Apologies for the noise. It seems to be a Pyright error.

closed 2024-06-26 by Yoric