pasqal-io/Pulser #702: As far as I understand, the QUBO/QAA tutorial is partially false

Yoric, 2024-07-01:
If I read the tutorial correctly:

1. The tutorial entirely ignores the diagonal terms.
2. The tutorial entirely ignores the case in which a non-diagonal term is <0.
3. The tutorial can produce layouts with atoms that are too close to each other or too far from the center.

HGSilveri, 2024-07-01:
Thanks for your comments @Yoric. I completely agree that this tutorial needs work. However, I have some clarifications to add to each of the points you raised:

> 1. The tutorial ignores the diagonal terms.

If by "ignores" you mean that they are not included in the optimization, I have to disagree. While it is true that only the off-diagonal terms are considered when trying to optimize the positions of the atoms, this is because the diagonal terms don't depend on the interaction between atoms but rather on the state of the atoms themselves. Put another way, the placement of the atoms has no influence on the diagonal terms.
Instead, they depend on the detuning term of the Hamiltonian, which you'll find is intentionally positive to energetically favor atoms to be in the $|1\rangle$ state (remember that the detuning appears as $-\delta$ in the Hamiltonian).
I agree that this is not being explained so we **will improve this**, but the diagonal terms are still there.

> 2\. The tutorial ignores the case in which a non-diagonal term is <0.

Indeed, no reference is made to the case when the non-diagonal terms are negative. In our defense, we are not claiming to have a universal QUBO solver, just showing how to solve a specific instance of one. That being said, I do see value in explicitly stating the limitations of the adopted approach, so we shall add this too.

> 3\. The tutorial can produce layouts with atoms that are too close to each other or too far from the center.

I would say these are device-dependent considerations that we are not going into for the sake of simplicity. The very simple approach for register embedding is quite limited but again, we are not claiming it is general. In fact, finding the optimal register embedding for a generic QUBO is a very hard problem that is decidedly outside the scope of this tutorial. Nonetheless, we can still highlight these limitations to not give the wrong impression, so let's do that too.

Yoric, 2024-07-01:
To clarify, I don't mean that the _algorithm_ ignores these terms, I mean that the _explanation_ ignores them (I realize that the term "ignore" is probably ambiguous, my bad). Having read this tutorial carefully, I have not reached an understanding of how we deal with 1. or 2.

> In our defense, we are not claiming to have a universal QUBO solver, just showing how to solve a specific instance of one.

Indeed, but as a reader of this tutorial, I (mistakenly) assumed that the tutorial was demonstrating a universal technique, so I feel that clarifying this point would be useful for future readers.

PR #731 (2024-09-24): Clarifications on the QUBO tutorial
Closes #702 .
closed 2024-09-24 by HGSilveri